belyalov/stm32-hal-libraries #1: Checksum

Guilyx, 2019-05-20:
Hey,

I'm currently using your Si7021 driver implementation on a STM32F4Discovery and I was wondering why, to measure humidity and temperature, you were using

`res = HAL_I2C_Master_Receive(hi2c, SI7021_ADDRESS_READ, si7021_buf, 3, 100);`

Why are you reading 3 bytes ? Are you reading the checksum too ? 
The doc says the checksum is optional, so 2 bytes should work as well right ?

Thanks.

belyalov, 2019-05-20:
Hey @Guilyx 

Yeah, you're right, 2 bytes should be enough.

I was going to implement checksum verification as well, but, ended up with not to do that because:
1. Not all chips support hardware CRC / or it maybe disabled in particular project.
2. Using software implementation will increase byte footprint.
3. Implementing another function like `si7021_measure_temperature_crc()`, however, looks like nobody will use it.

So feel free to clean it up :) (or I'll clean it up in the future)

Guilyx, 2019-05-21:
Hey,

I opened up the merge request. I had another question, saw that on multiple drivers : 
`#define SI7021_ADDRESS_READ          (0x40 << 1) | 0x01`
`#define SI7021_ADDRESS_WRITE         (0x40 << 1)`

Why do you apply a left offset on the base address ?
Thought the read/write command was on the last byte of the device's address, not on the last byte of the 8 bits word ? 
I'm not sure at all, because everyone made their driver this way, I'm just asking.

Documentation from STM32 I2C says :
> To enter Transmitter mode, a master sends the slave address with LSB reset.
> To enter Receiver mode, a master sends the slave address with LSB set. 

Any clue/explanations on that please ?

Thanks, I appreciate it!

belyalov, 2019-05-21:
This is common pattern to interact through `I2C` bus:
[image]
- to perform **write** operation on remote `i2c` device set first bit to `0` (actually just shift left)
- to perform **read** - set it to `1`

Guilyx, 2019-05-22:
Alright. Thanks alot.

belyalov, 2019-05-22:
Fixed by #2

closed 2019-05-22 by belyalov