belyalov/stm32-hal-libraries #3: About LoRa and NSS

selimsagir, 2020-11-20:
Hi, 

I am using B-L072Z-LRWAN1 kit including CMWX1ZZABZ-091 module. You mentioned that;

`  // SX1276 compatible module connected to SPI1, NSS pin connected to GPIO with label LORA_NSS
  uint8_t res = lora_init(&lora, &hspi1, LORA_NSS_GPIO_Port, LORA_NSS_Pin, LORA_BASE_FREQUENCY_US);
  if (res != LORA_OK) {
    // Initialization failed
  }`

But after I've created a new project with CubeMx for B-L072Z-LRWAN1 kit it shows Hardware NSS Signal is disable by default and when I check code it shown as **SPI_NSS_SOFT.**  at init struct.

So what should I do? I am a bit confused about that.

Thank you.

belyalov, 2020-11-20:
Hey,

This example does not use hardware NSS, just any pin configured as `GPIO_OUTPUT` will work.
Something like
[image]

selimsagir, 2020-11-21:
Thank you for your help. But it does not work still. Should I set some parameters, such as SF, tx_power etc?

belyalov, 2020-11-22:
It is good to have them setup explicitly on all involved devices - at least to be sure that your devices operates on the same SF/FREQ.

belyalov, 2020-12-30:
Please feel free to open new issue if you need advice.. )

Closing this one as per no activity.

closed 2020-12-30 by belyalov

mamanbudiman, 2021-01-29:
Hello @belyalov 
I am trying to use your lora library using stm32cubeIDE on STM32L151C8 MCU.
But, still doesn't worked. 

`
  lora_sx1276 lora;

   // SX1276 compatible module connected to SPI1, NSS pin connected to GPIO with label LORA_NSS
   uint8_t res = lora_init(&lora, &hspi1, LORA_NSS_GPIO_Port, LORA_NSS_Pin, LORA_BASE_FREQUENCY_US);
   if (res != LORA_OK) {
     // Initialization failed
	   printf("init failed \n\r");
   }

  /* USER CODE END 2 */

  /* Infinite loop */
  /* USER CODE BEGIN WHILE */
  while (1)
  {
    /* USER CODE END WHILE */
//	  HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_RESET);  // power down sensor
//	  HAL_Delay(1000);
//	  HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_SET);  // power down sensor
//	  HAL_Delay(1000);
	  counter++;
	  char bufData[60];
	  sprintf(bufData, "%d \n\r", counter);
	  printf(bufData);

	  // Send packet can be as simple as
//	  uint8_t res = lora_send_packet(&lora, (uint8_t *)"test", 4);
//	  if (res != LORA_OK) {
//	    // Send failed
//		  printf("send failed \n\r");
//	  } else {
//		  printf("send ok \n\r");
//	  }

	  // Receive buffer
	    uint8_t buffer[32];
	    // Put LoRa modem into continuous receive mode
	    //lora_mode_receive_continuous(&lora);
	    lora_mode_receive_single(&lora);

	    // Wait for packet up to 10sec
	    uint8_t res;
	    uint8_t len = lora_receive_packet_blocking(&lora, buffer, sizeof(buffer), 1000, &res);
	    //uint8_t len = lora_receive_packet(&lora, buffer, sizeof(buffer), &res);
	    if (res != LORA_OK) {
	      printf("received failed \n\r");
	    }
	    buffer[len] = 0;  // null terminate string to print it
	    printf("'%s'\n", buffer);

    /* USER CODE BEGIN 3 */
  }
  /* USER CODE END 3 */
}
`
Is there any other requirements that probably I miss it.
Do you have an example simple project that using your lora library ?

Thank you for your help.

belyalov, 2021-01-30:
It is hard to say what is wrong, but, I have seen few problems with transmitting packets over lora:
- Before `lora_send_packet(...)` please be sure to put module into standby mode by `lora_mode_standby()`
- Check module power supply: make sure that you have bypass capacitors (100nf and 1uf recommended).

Also, during transmission there will be power consumption spikes up to 20mA (maybe even more, depending on environment and configuration) so be sure that your power regulator (if any) can provide such power.

mamanbudiman, 2021-01-30:
Hello @belyalov 

Thanks for your reply. 
I have added lora_mode_standby(), but still not working.
Is there any SPI parameters that should be set.
`static void MX_SPI1_Init(void)
{
  /* USER CODE BEGIN SPI1_Init 0 */
  /* USER CODE END SPI1_Init 0 */
  /* USER CODE BEGIN SPI1_Init 1 */
  /* USER CODE END SPI1_Init 1 */
  /* SPI1 parameter configuration*/
  hspi1.Instance = SPI1;
  hspi1.Init.Mode = SPI_MODE_MASTER;
  hspi1.Init.Direction = SPI_DIRECTION_2LINES;
  hspi1.Init.DataSize = SPI_DATASIZE_8BIT;
  hspi1.Init.CLKPolarity = SPI_POLARITY_LOW;
  hspi1.Init.CLKPhase = SPI_PHASE_1EDGE;
  hspi1.Init.NSS = SPI_NSS_SOFT;
  hspi1.Init.BaudRatePrescaler = SPI_BAUDRATEPRESCALER_64;
  hspi1.Init.FirstBit = SPI_FIRSTBIT_MSB;
  hspi1.Init.TIMode = SPI_TIMODE_DISABLE;
  hspi1.Init.CRCCalculation = SPI_CRCCALCULATION_DISABLE;
  hspi1.Init.CRCPolynomial = 10;
  if (HAL_SPI_Init(&hspi1) != HAL_OK)
  {
    Error_Handler();
  }
  /* USER CODE BEGIN SPI1_Init 2 */

  /* USER CODE END SPI1_Init 2 */

}
`
Best regards,

belyalov, 2021-01-30:
Looks good, this is from my actually working project:
```c
  hspi1.Instance = SPI1;
  hspi1.Init.Mode = SPI_MODE_MASTER;
  hspi1.Init.Direction = SPI_DIRECTION_2LINES;
  hspi1.Init.DataSize = SPI_DATASIZE_8BIT;
  hspi1.Init.CLKPolarity = SPI_POLARITY_LOW;
  hspi1.Init.CLKPhase = SPI_PHASE_1EDGE;
  hspi1.Init.NSS = SPI_NSS_SOFT;
  hspi1.Init.BaudRatePrescaler = SPI_BAUDRATEPRESCALER_2;
  hspi1.Init.FirstBit = SPI_FIRSTBIT_MSB;
  hspi1.Init.TIMode = SPI_TIMODE_DISABLE;
  hspi1.Init.CRCCalculation = SPI_CRCCALCULATION_DISABLE;
  hspi1.Init.CRCPolynomial = 7;
```

Also, check connection of RST pin - maybe it is accidentally pulled down?

mamanbudiman, 2021-01-30:
Hello @belyalov 

Finally, it is working.
Thanks for your support.

Regards,

belyalov, 2021-01-30:
Out of curiosity: what was the problem?)

mamanbudiman, 2021-01-30:
I set the SPI parameters as above and explicitly set the RST pin of the LORA module.
Previously, I use Radiolib library, they have RST pin on Lora init function.

Regards,

ronandunn, 2022-01-13:
A couple of problems that I had using the B-L072z-LRWAN1 board were needing to reset the radio before initializing (pin C0) and making sure LORA_NSS was pin A15.

francescoNoto, 2024-03-05:
Hi, what does it mean to explicitly set the rst of the module? could you explain to me please? Thank you