belyalov/stm32-hal-libraries #5: LoRa version problem

pyrotechnik, 2021-06-30:
Hello, I would like to use your library for SX1276. But I have a problem with verifying the version of the module. I get an answer of 0 and if I switch the MISO with MOSI I will get 255. I have already tried a different connections and I use reset LoRa modul using the RST pin, but still without results.

xgonc, 2021-09-24:
@pyrotechnik I'm having the same issue. Did you solve this? Can you share?

pyrotechnik, 2021-09-24:
Hello, unfortunately not.  I bought an SX1278.  And here everything works properly.

xgonc, 2021-09-24:
Meanwhile I figured out what my problem was: the SPI was configured at 40MHz and 4 bits. I changed to 10MHz and 8 bits and started working.

belyalov, 2021-11-16:
Great that you've found the problem yourself.
BTW, I recently fit the same stuff: for some reason CubeMX set non default SPI params, so always great to double check whatever is being generated by cubemx.

Closing this one as resolved

closed 2021-11-16 by belyalov